platformio/platformio-atom-ide #509: Problem occured while uprading PlatformIO IDE.

scopa90, 2017-05-26:
Error: Unable to download https://www.atom.io/api/packages/platformio-ide/versions/2.0.0-beta.4/tarball: 503

    at utils.runAPMCommand.busyTitle (file:///C:/Users/scopa/.atom/packages/platformio-ide/lib/installer/ide.js:111:13)
    at onExit (file:///C:/Users/scopa/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/scopa/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\scopa\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:338:1)
    at ChildProcess.<anonymous> (C:\Users\scopa\AppData\Local\atom\app-1.17.0\resources\app\src\buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

scopa90, 2017-05-26:
Whilst browsing a folder looking for an existing project.

ivankravets, 2017-05-26:
It looks like GitHub has a problem... See 503 error

closed 2017-05-26 by ivankravets