platformio/platformio-atom-ide #510: C/C++ project index rebuild failed

sjschaff, 2017-05-26:
I moved some project library files from outside of atom (specifically moved lib/glm -> lib/glm/glm due to glm's internal header usages), and I immediately started getting this error from platformio.  Deleting .pioenvs fixed it, but presumably shouldn't have happened

Exception: [Errno 13] Permission denied: '/Users/<user>/Files/Projects/Totem/platformio/.pioenvs/do-not-modify-files-here.url':

File "/Users/<user>/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 86:
PROJECTPIOENVS_DIR=util.get_projectpioenvs_dir(),
File "/Users/<user>/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 309:
raise Exception(e)

========================== [ERROR] Took 5.14 seconds ==========================

ivankravets, 2017-05-26:
Just close Atom, remove `/Users//Files/Projects/Totem/platformio/.pioenvs` and start Atom

closed 2017-05-26 by ivankravets

sjschaff, 2017-05-26:
Yeah I got it fixed, thanks.  Still seems like a bug though.

ivankravets, 2017-05-26:
Something has gone with permission :(