platformio/platformio-atom-ide #511: C/C++ project index rebuild failed

sembsa, 2017-05-27:
Traceback (most recent call last):
  File "/Users/sembsa/.platformio/penv/bin/platformio", line 6, in <module>

  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3036, in <module>

    @_call_aside
  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3020, in _call_aside
    f(*args, **kwargs)
  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3049, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 654, in _build_master
    ws.require(__requires__)
  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 968, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Users/sembsa/.platformio/penv/lib/python2.7/site-packages/pkg_resources/__init__.py", line 854, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==3.4.0b3' distribution was not found and is required by the application

ivankravets, 2017-05-27:
Please remove `/Users/sembsa/.platformio/penv/` and restart Atom

ivankravets, 2017-06-01:
Please reopen if you still need help

closed 2017-06-01 by ivankravets

No3x, 2017-06-24:
Deleting the penv dir worked for me.