platformio/platformio-atom-ide #512: C/C++ project index rebuild failed

wjm0621, 2017-05-27:
C:\Users\AI_Lab\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
val = [x for x in val if x not in dk]

UnicodeDecodeError: 'utf8' codec can't decode byte 0xc1 in position 24: invalid start byte:
File "C:\Users\AI_Lab\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

print "\n%s\n" % json.dumps(env.DumpIDEData())
File "C:\Users\AI_Lab\.platformio\python27\Lib\json\__init__.py", line 244:
return _default_encoder.encode(obj)
File "C:\Users\AI_Lab\.platformio\python27\Lib\json\encoder.py", line 207:
chunks = self.iterencode(o, _one_shot=True)
File "C:\Users\AI_Lab\.platformio\python27\Lib\json\encoder.py", line 270:
return _iterencode(o, 0)

 [ERROR] Took 4.75 seconds

ivankravets, 2017-05-27:
Could you share `platformio.ini` contents here?

wjm0621, 2017-05-27:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32thing]
platform = espressif32
board = esp32thing
framework = arduino

ivankravets, 2017-05-27:
Do you have Non-Unicode chars in file names or inside source code?

wjm0621, 2017-05-27:
Hmm.... I Just created new project and i saw this problem.
I reinstalled Atom and platformIO IDE. but this is continued when i first installed PlatformIO.
I'm using Windows 7(64bit). and my Windows User ID is AI_Lab.
I can't infer what is this problem..

wjm0621, 2017-05-27:
I'm finding about 0xC1.

Microsoft says this problem will be malware or registry damaged. or I'm using old driver.
and said install Windows update.
perhaps it's not old driver problem.

I install all updates for Windows, failed one.
I executed sfc /scannow at the cmd (administrator autority.)
I deleted temporary folders.

But this error continued.

ivankravets, 2017-05-27:
Please mail me@ikravets.com for the further instructions. Thanks!

ivankravets, 2017-06-01:
See http://docs.platformio.org/en/latest/faq.html#unicodedecodeerror-non-ascii-characters-found-in-build-environment

closed 2017-06-01 by ivankravets