platformio/platformio-atom-ide #513: C/C++ project index rebuild failed

vbw-001, 2017-05-27:
Error: spawn platformio ENOENT

Endless restart loop - spawns several of these errors, then asks to restart, and repeat.

Tried reinstall of atom, platformio packages, no luck

ivankravets, 2017-05-27:
Could you run `pio update` in PIO IDE Terminal and restart Atom?

vbw-001, 2017-05-27:
bash: pio: command not found

Can't even get to installing the shell commands from the menu - most of the items in the PlatformIO menu do nothing at all (no error, just nothing).

Restart just continues to re-trigger the install. Weird thing is I didn't do anything to set this off as far I can tell - just rebooted my machine and tried to start atom.

ivankravets, 2017-05-27:
Please mail me@ikravets.com for detailed instructions. Thanks!

vbw-001, 2017-05-27:
Thanks to Ivan for the help, a careful re-install of all components successfully fixed the issue. My fault for missing one of the components the first time I tried to uninstall.

closed 2017-05-27 by vbw-001