platformio/platformio-atom-ide #514: after it says installed on windows 10

ridua, 2017-05-27:
Cannot find module 'C:\Users\Rishabbh\.atom\packages\platformio-ide\node_modules\extract-opts\package.json'
Hide Stack Trace
Error: Cannot find module 'C:\Users\Rishabbh\.atom\packages\platformio-ide\node_modules\extract-opts\package.json'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\electron.asar\common\reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\module-cache.js:354:1)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:40:43)
    at require (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:66:1)
    at Object.requirePackage (C:\Users\Rishabbh\.atom\packages\platformio-ide\node_modules\editions\es2015\index.js:211:17)
    at Object.<anonymous> (C:\Users\Rishabbh\.atom\packages\platformio-ide\node_modules\extract-opts\index.js:3:38)
    at Module.get_Module._compile (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:106:1)
    at Object.value [as .js] (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:66:1)
    at Object.<anonymous> (C:\Users\Rishabbh\.atom\packages\platformio-ide\node_modules\getmac\es5\lib\getmac.js:6:26)
    at Module.get_Module._compile (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:106:1)
    at Object.value [as .js] (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:66:1)
    at Object.<anonymous> (C:\Users\Rishabbh\.atom\packages\platformio-ide\lib\telemetry.js:29:15)
    at Module.get_Module._compile (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:106:1)
    at Object.value [as .js] (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:66:1)
    at Object.<anonymous> (C:\Users\Rishabbh\.atom\packages\platformio-ide\lib\maintenance.js:34:18)
    at Module.get_Module._compile (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:106:1)
    at Object.value [as .js] (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:66:1)
    at Object.<anonymous> (C:\Users\Rishabbh\.atom\packages\platformio-ide\lib\main.js:33:20)
    at Module.get_Module._compile (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\native-compile-cache.js:106:1)
    at Object.value [as .js] (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/Rishabbh/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\static\<embedded>:73:1)
    at Package.module.exports.Package.requireMainModule (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\package.js:887:1)
    at Package.module.exports.Package.activateNow (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\package.js:242:1)
    at C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\package.js:1014:1
    at Function.module.exports.Emitter.simpleDispatch (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\node_modules\event-kit\lib\emitter.js:25:1)
    at Emitter.module.exports.Emitter.emit (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\node_modules\event-kit\lib\emitter.js:141:1)
    at C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\package-manager.js:655:1
    at Set.forEach (native)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\package-manager.js:653:1)
    at C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\node_modules\settings-view\lib\package-manager.js:519:1
    at exit (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\node_modules\settings-view\lib\package-manager.js:128:1)
    at triggerExitCallback (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:338:1)
    at ChildProcess.<anonymous> (C:\Users\Rishabbh\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

ivankravets, 2017-05-27:
Could you uninstall `platformio-ide` package and install it again?

ivankravets, 2017-06-01:
Beta5 is soon

closed 2017-06-01 by ivankravets