platformio/platformio-atom-ide #515: C/C++ project index rebuild failed

bschick, 2017-05-27:
Error: spawn platformio ENOENT

Updated to the latest packages on OSX and platformio update is now stuck in a loop. I see several instances of the error above while "core" is being updated, followed by a success message, and reboot button. Rebooting starts the process over again and platformio ide is never installed correctly (menu options do nothing)

ivankravets, 2017-05-27:
Could you try to remove `~/.platformio/penv` folder and retsart Atom?

bschick, 2017-05-27:
Thanks much, that solved it. I still got one C++ rebuild Error: spawn platformio ENOENT error and reboot but after that PIO is installed and working.

closed 2017-05-27 by bschick

ivankravets, 2017-05-27:
1. http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide
2. http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Please remove all PIO Cores from a system and make clean installation.