platformio/platformio-atom-ide #519: InstallationManager

electromc2, 2017-05-27:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-4438XFORqaEKTXFx/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()

  File "/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-4438XFORqaEKTXFx/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-4438XFORqaEKTXFx/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/terry/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/terry/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 54, in isabs

    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-05-27:
Could you remove `/Users/terry/.platformio/penv` and restart Atom? Did it help?

electromc2, 2017-05-27:
Ivan I tried your suggestion.  Every time  I start Atom/Platformio IDE 
It upadate and seems to crash while updating

Plaformio core.

Thanks   Terry

Virtualenv Download: Traceback (most recent call last):
File 
"/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-8485Sy9uV7el4Cet/virtualenv-14.0.6/virtualenv.py", 
line 2304, in <module>
main()
File 
"/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-8485Sy9uV7el4Cet/virtualenv-14.0.6/virtualenv.py", 
line 705, in main
symlink=options.symlink)
File 
"/var/folders/qk/d0799f_91yq__qchf1xq3npc0000gn/T/tmp-8485Sy9uV7el4Cet/virtualenv-14.0.6/virtualenv.py", 
line 904, in create_environment
site_packages=site_packages, clear=clear, symlink=symlink))
File "/Users/terry/.platformio/penv/bin/../lib/python2.7/posixpath.py", 
line 360, in abspath
if not isabs(path):
File "/Users/terry/.platformio/penv/bin/../lib/python2.7/posixpath.py", 
line 54, in isabs
return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

electromc2, 2017-05-27:
Ivan I rebboted my mac and pip uninstalled requests and reinstalled. 
That seems to have fixed the problem.

Terry

ivankravets, 2017-06-01:
Thanks for the report!

closed 2017-06-01 by ivankravets