platformio/platformio-atom-ide #520: InstallationManager

VladUrziceanu, 2017-05-27:
Conda Virtualenv: /bin/sh: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

ERROR conda.core.link:_execute_actions(335): An error occurred while installing package 'defaults::readline-6.2-2'.
CondaError: Cannot link a source that does not exist. /opt/anaconda/pkgs/readline-6.2-2/lib/libhistory.a

Attempting to roll back.

CondaError: Cannot link a source that does not exist. /opt/anaconda/pkgs/readline-6.2-2/lib/libhistory.a

ivankravets, 2017-05-27:
Do you use the latest Conda?

VladUrziceanu, 2017-05-27:
Yes, i just did an update.
conda 4.3.21

ivankravets, 2017-05-27:
Could you remove `~/.platformio/penv` folder and restart Atom?

VladUrziceanu, 2017-05-28:
I still get the same error. I also tried removing `platformio-ide` package from atom and reinstalling.

VladUrziceanu, 2017-06-04:
It was an anaconda problem. Fixed by running 
```
conda remove readline -y --force
conda clean --all -y
conda install readline -y
```
The were also other broken packages. Just run the same thing for all of them.

closed 2017-06-04 by VladUrziceanu