platformio/platformio-atom-ide #521: After install PlatformIO-IDE on Atom, Atom show restart Window Infinitly

wjm0621, 2017-05-27:
[image]
[image]

after remove atom and platformio, LLVM, Python 2.7 I reinstalled Atom and platformIO-IDE.
but Atom just show Message "please restart Atom to apply the changes"
after restart, show again
after restart, show again....
....
infinitly

how can i solve this problem?

ivankravets, 2017-05-27:
https://github.com/platformio/platformio-core/releases/tag/v3.3.1

Please restart Atom

closed 2017-05-27 by ivankravets

wjm0621, 2017-05-27:
Thank you ivankravets!! I solved this problem. Thank you a million!!!