platformio/platformio-atom-ide #523: Uncaught TypeError: Cannot read property 'doRebuild' of undefined

wjm0621, 2017-05-27:
[Enter steps to reproduce:]

1. Open 2 projects, the first project is broken. Rebuild C/C++ reports an error....
2. Now, try to rebuild normal project (2-nd)

**Atom**: 1.17.2 x64
**Electron**: 1.3.15
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.4

### Stack Trace

Uncaught TypeError: Cannot read property 'doRebuild' of undefined

```
At C:\Users\AI_Lab\.atom\packages\platformio-ide\lib\project\observer.js:83

TypeError: Cannot read property 'doRebuild' of undefined
    at ProjectObserver.rebuildActiveProject (/packages/platformio-ide/lib/project/observer.js:83:61)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.17.2/resources/app/src/command-registry.js:265:1)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.17.2/resources/app/src/command-registry.js:3:1)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (~/AppData/Local/atom/app-1.17.2/resources/app/src/command-registry.js:166:1)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (~/AppData/Local/atom/app-1.17.2/resources/app/src/atom-environment.js:1300:1)
    at EventEmitter.outerCallback (~/AppData/Local/atom/app-1.17.2/resources/app/src/application-delegate.js:334:1)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
     -3:55 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
     -3:09.0 window:toggle-dev-tools (input.hidden-input)
     -2:47.9.0 platformio-ide:maintenance.rebuild-index (input.hidden-input)
     -2:08.4.0 platformio-ide:home (input.hidden-input)
     -1:46.2.0 tree-view:remove-project-folder (span.name.icon.icon-file-directory)
  2x -1:46.2.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:41.8.0 platformio-ide:maintenance.rebuild-index (input.hidden-input)
     -1:31.8.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:28.9.0 tree-view:remove-project-folder (div.header.list-item.project-root-header)
  2x -1:28.8.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:03.2.0 tree-view:remove-project-folder (span.name.icon.icon-file-directory)
     -1:03.2.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:56.9.0 platformio-ide:maintenance.rebuild-index (input.hidden-input)
  2x -0:11.9.0 core:backspace (input.hidden-input)
     -0:08.0 intentions:highlight (input.hidden-input)
     -0:07.8.0 core:save (input.hidden-input)
```

### Non-Core Packages

```
autocomplete-clang 0.11.3 
build 0.68.0 
busy 0.7.0 
busy-signal 1.4.1 
file-icons 2.1.6 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.1.4 
linter-gcc 0.7.1 
linter-ui-default 1.6.0 
minimap 4.28.2 
platformio-ide 2.0.0-beta.4 
platformio-ide-debugger 1.2.2 
platformio-ide-terminal 2.5.1 
tool-bar 1.1.0 
```

ivankravets, 2017-05-27:
@orgkhnargh please fix

ardentsword, 2017-06-01:
Still happens after closing the first (broken) project. Restarting the IDE with only the second (working) project 'fixes' the issue.

atar-axis, 2017-09-21:
for me a 'clean' before 'rebuild' worked ;)

SRGDamia1, 2017-10-23:
Is there any fix for this?

ivankravets, 2017-10-23:
@SRGDamia1 could you try to uninstall `platformio-ide` extension and install it again?

SRGDamia1, 2017-10-24:
Uninstalling and reinstalling didn't make any difference for me.

vk5zm, 2017-10-26:
I have the same issue with my PIO install and ditto, re-installing did not fix the issue either.

ivankravets, 2017-11-04:
@SRGDamia1 do you experience this issue with new 2.0.0-rc.4 update?

SRGDamia1, 2017-11-06:
Yes, I'm still getting it.  I have platformio-ide 2.0.0-rc.4 with platformio 3.5.0b2 on 64-bit Windows 7.  I just (today) uninstalled the ide package within atom, deleted the remaining .platformio folder in my user folder, and then reinstalled and I still get the error.

The PlatformIO Home screen seems to be loading more quickly and consistently, though!

SRGDamia1, 2017-11-06:
Oops.  I'm sorry.  I have no idea how I unassigned @KyrychukD.

flurl, 2017-11-30:
I had the same problem. Any chance that there's a symbolic link in the path to your project folder? That's been the problem for me (once again with Atom…).
When I add the project folder via its original path it works.

closed 2017-12-28 by ivankravets

SRGDamia1, 2018-01-10:
I'm still having this problem, with every project every time.

ivankravets, 2018-01-10:
@SRGDamia1 could you contact me@ikravets.com? Thanks!

SRGDamia1, 2018-01-12:
After completely and thoroughly removing PlatformIO, Atom, and Anaconda/Python and then reinstalling them all, this finally seems to be fixed for me.  I'll let you know if it comes up again.

Thank you!

ivankravets, 2018-01-12:
@SRGDamia1 Oh... You have used Anaconda... I only yesterday fixed a critical bug... Now I understand why it doesn't work for you before and works now. Sorry.

SRGDamia1, 2018-01-12:
I'm sorry, I should have specified.  Yes, I use Anaconda for everything python.  The conda package management is great, especially when I don't want to have to keep up with all the updates to NumPy/Pandas/etc.

vk5zm, 2018-01-12:
@ivankravets your recent bug fix has also resolved the same issue for me without having to reinstall, many thanks for your hardwork !

commit 13a68905 (2017-12-28): Fix issue with symlink and project index rebuilding // Resolve #523