platformio/platformio-atom-ide #524: InstallationManager

chiduc80, 2017-05-27:
Virtualenv Download: Traceback (most recent call last):
  File "F:\Temp\tmp-8520BxbYhEM52Axi\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "F:\Temp\tmp-8520BxbYhEM52Axi\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "F:\Temp\tmp-8520BxbYhEM52Axi\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "F:\Temp\tmp-8520BxbYhEM52Axi\virtualenv-15.1.0\virtualenv.py", line 848, in install_wheel
    from urllib.parse import urljoin
  File "C:\Python27\lib\urllib.py", line 26, in <module>
    import socket
  File "C:\Python27\lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: %1 is not a valid Win32 application.

ivankravets, 2017-06-01:
I see that you don't use the latest PIO IDE Installer. Could you reinstall with http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide ?

ivankravets, 2017-06-27:
Please reopen if you still need help

closed 2017-06-27 by ivankravets

Msiavashi, 2017-09-30:
i've got the same problem . both on atom and vscode .