platformio/platformio-atom-ide #533: Uncaught SyntaxError: Unexpected token A in JSON at position 0

vokom, 2017-05-28:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.17.2 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.11.5
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.4

### Stack Trace

Uncaught SyntaxError: Unexpected token A in JSON at position 0

```
At file:///Users/vokom/software/dev-tools/Atom.app/Contents/Resources/app/static/index.html:1

SyntaxError: Unexpected token A in JSON at position 0
    at Object.parse (native)
    at /packages/platformio-ide/lib/account/helpers.js:29:55
    at onExit (/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Users/vokom/software/dev-tools/Atom.app/Contents/Resources/app/src/buffered-process.js:338:1)
    at /Users/vokom/software/dev-tools/Atom.app/Contents/Resources/app/src/buffered-process.js:351:1
    at Socket.<anonymous> (/Users/vokom/software/dev-tools/Atom.app/Contents/Resources/app/src/buffered-process.js:220:1)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
     -0:17.6.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
```

### Non-Core Packages

```
atom-hide-tabs 0.3.1 
autocomplete-clang 0.11.3 
autocomplete-cmake 0.3.1 
build 0.67.0 
build-cmake 0.9.1 
busy 0.7.0 
change-tabs 0.2.0 
file-icons 2.1.2 
gcc-make-run 0.2.9 
intentions 1.1.2 
language-cmake 0.3.0 
language-ini 1.19.0 
linter 2.1.2 
linter-gcc 0.7.1 
linter-ui-default 1.2.2 
minimap 4.27.0 
platformio-ide 2.0.0-beta.4 
platformio-ide-debugger 1.2.2 
platformio-ide-terminal 2.5.1 
tidy-tabs 0.4.0 
tool-bar 1.1.0 
```

ivankravets, 2017-06-01:
Could you reproduce this issue after restart?

ivankravets, 2017-06-03:
Please reopen if you need a help

closed 2017-06-03 by ivankravets