platformio/platformio-atom-ide #537: Can't find "async" after upgrade to PIO 2.0.0-b4

sburlappp, 2017-05-29:
[Enter steps to reproduce:]

1. Update All
2. Much is broken after restart

**Atom**: 1.17.2 ia32
**Electron**: 1.3.15
**OS**: Microsoft Windows 10 Home
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.4

### Stack Trace

Failed to activate the platformio-ide package

```
At Cannot find module 'async'

Error: Cannot find module 'async'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (~/AppData/Local/atom/app-1.17.2/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (~/AppData/Local/atom/app-1.17.2/resources/app/src/module-cache.js:354:1)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:40:43)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at /packages/platformio-ide/node_modules/fs-plus/lib/fs-plus.js:13:11)
    at /packages/platformio-ide/node_modules/fs-plus/lib/fs-plus.js:718:4)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at /packages/platformio-ide/lib/config.js:17:15)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at /packages/platformio-ide/lib/main.js:25:15)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
```

### Commands

```
     -1:50.3.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
```

### Non-Core Packages

```
autocomplete-clang 0.11.3 
build 0.68.0 
busy 0.7.0 
file-icons 2.1.6 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.1.4 
linter-gcc 0.7.1 
linter-ui-default 1.6.0 
minimap 4.28.2 
platformio-ide 2.0.0-beta.4 
platformio-ide-debugger 1.2.2 
platformio-ide-terminal 2.5.1 
tool-bar 1.1.0 
```

ivankravets, 2017-06-01:
Could you try uninstall/install PIO IDE?

closed 2017-06-01 by ivankravets

sburlappp, 2017-06-02:
Fixed it, thanks.