platformio/platformio-atom-ide #539: InstallationManager

piodabro, 2017-05-29:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 780, in call_subprocess
    logger.notify('\n'.join(all_output) + '\n----------------------------------------')
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 193, in notify
    self.log(self.NOTIFY, msg, *args, **kw)
  File "C:\Users\PDABRO~1.GPX\AppData\Local\Temp\tmp-18472XPJ6dP4oh7sQ\virtualenv-14.0.6\virtualenv.py", line 225, in log
    consumer.write(rendered+'\n')
UnicodeEncodeError: 'ascii' codec can't encode character u'\u0119' in position 2916: ordinal not in range(128)

piodabro, 2017-05-29:
This happened when updated to 2.0.0.-beta.4.
Also clean Atom installation shows the same.
Sys: Windows 10

ivankravets, 2017-06-01:
Try to remove `~/.platformio` folder and start again

closed 2017-06-01 by ivankravets