platformio/platformio-atom-ide #540: C/C++ project index rebuild failed

jalips, 2017-05-29:
Error: Traceback (most recent call last):
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 74, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/maintenance.py", line 68, in on_platformio_start
    after_upgrade(ctx)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/maintenance.py", line 189, in after_upgrade
    update_core_packages(silent=True)
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/core.py", line 61, in update_core_packages
    if not silent or pm.outdated(pkg_dir, requirements):
  File "/home/link/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/package.py", line 554, in outdated
    assert isdir(pkg_dir)
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-06-01:
Could you provide an output of  `pio platform list`?

ivankravets, 2017-06-01:
Thanks, fixed in https://github.com/platformio/platformio-core/commit/5da5bd43e4138ecef0b001cb312138e55d73c2ec

closed 2017-06-01 by ivankravets

jalips, 2017-06-01:
Too quick ! nice job

ivankravets, 2017-06-01:
Beta 5 of PIO IDE will be released soon