platformio/platformio-atom-ide #543: Error installing espressif8266_stage - macOS Sierra

raphaelwb, 2017-05-31:
Error: Could not install 'framework-arduinoespressif8266' with version requirements '1.20400.0-alpha.2' for your system 'darwin_x86_64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

ivankravets, 2017-05-31:
We are working on moving of PlatformIO Build Logic to an official ESP8266 repo.

Could you try again?

raphaelwb, 2017-05-31:
Hello,

Thanks for the quick response. The error persists. I'm trying to use the ESPAsyncUDP library and at compile time the following error occurs: "... udp_set_multicast_netif_addr 'was not declared ...". I understand that is necessary to update the arduino core version, so I used the instructions at: http://docs.platformio.org/en/stable/platforms/espressif8266.html#using-arduino-framework-with-staging-version. The error occurs when compiling using the espressif8266_stage platform:

Platform = espressif8266_stage
Board = d1_mini
Framework = arduino

I have the last version of plataformIO updated and the operational system is a macOS Sierra 10.12.4

Thank you

ivankravets, 2017-05-31:
@valeros working on this error.

valeros, 2017-05-31:
Hi @raphaelwb ! Please update platform, it should be fixed now.

raphaelwb, 2017-05-31:
Tks. it is ok now ...

closed 2017-05-31 by raphaelwb