platformio/platformio-atom-ide #551: Platform command failed: platform search --json-output

dude719, 2017-06-01:
This seems to be completely broken on Windows. Getting tons of errors right after installation.

```
Error: Error: Could not load broken JSON: C:\Users\Owner\.platformio\appstate.json

    at file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/platform/helpers.js:61:25
    at onExit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:338:1)
    at ChildProcess.<anonymous> (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

```
Error: Error: Could not load broken JSON: C:\Users\Owner\.platformio\appstate.json

    at file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/platform/helpers.js:61:25
    at onExit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:338:1)
    at ChildProcess.<anonymous> (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

ivankravets, 2017-06-01:
Could you share content of this file `C:\Users\Owner\.platformio\appstate.json` here? Thanks!

dude719, 2017-06-01:
```
{
   "username":null,
   "cid":"6ea9fbac-0ce3-ca2e-549b-dbcb5e627f7f",
   "last_version":"3.4.0b9",
   "telemetry":{
      "backup":[

      ]
   },
   "last_check":{
      "platformio_upgrade":1496355887,
      "platforms_update":1496355888,
      "libraries_update":1496355888
   },
   "stoken":null
}
```

dude719, 2017-06-01:
I get this error along with it, sorry I didn't add it earlier:

```
Error: Traceback (most recent call last):
  File "C:\Users\Owner\.platformio\penv\Scripts\platformio-script.py", line 11, in <module>
    load_entry_point('platformio==3.4.0b9', 'console_scripts', 'platformio')()
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\__main__.py", line 107, in main
    maintenance.on_platformio_exception(e)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 87, in on_platformio_exception
    telemetry.on_exception(e)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 314, in on_exception
    mp = MeasurementProtocol()
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 65, in __init__
    self['cid'] = app.get_cid()
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 342, in get_cid
    cid = get_state_item("cid")
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 285, in get_state_item
    with State() as data:
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 89, in __enter__
    self._state = util.load_json(self.path)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\util.py", line 178, in load_json
    file_path)
platformio.exception.PlatformioException: Could not load broken JSON: C:\Users\Owner\.platformio\appstate.json
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Python27\Lib\atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 331, in _finalize
    backup_reports(MPDataPusher().get_items())
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 341, in backup_reports
    tm = app.get_state_item("telemetry", {})
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 285, in get_state_item
    with State() as data:
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 89, in __enter__
    self._state = util.load_json(self.path)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\util.py", line 178, in load_json
    file_path)
PlatformioException: Could not load broken JSON: C:\Users\Owner\.platformio\appstate.json
Error in sys.exitfunc:
Traceback (most recent call last):
  File "C:\Python27\Lib\atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 331, in _finalize
    backup_reports(MPDataPusher().get_items())
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 341, in backup_reports
    tm = app.get_state_item("telemetry", {})
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 285, in get_state_item
    with State() as data:
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\app.py", line 89, in __enter__
    self._state = util.load_json(self.path)
  File "c:\users\owner\.platformio\penv\lib\site-packages\platformio\util.py", line 178, in load_json
    file_path)
platformio.exception.PlatformioException: Could not load broken JSON: C:\Users\Owner\.platformio\appstate.json
    at file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/platform/helpers.js:61:25
    at onExit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/Owner/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:338:1)
    at C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:351:1
    at Socket.<anonymous> (C:\Users\Owner\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:220:1)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

dude719, 2017-06-01:
It happens when I try to access the Home->Platforms->Embedded or Desktop tabs.

ivankravets, 2017-06-01:
Something strange... could you remove this C:\Users\Owner\.platformio\appstate.json file?

dude719, 2017-06-01:
Okay, if I delete the appstate.json file and **dont** run as administrator, the problem is fixed. However, if I then run as administrator and go to Home->Platforms->Embedded or Desktop tabs the error pops up and _seemingly_ screws up the appstate.json file, but I don't see anything wrong with the appstate.json file.

ivankravets, 2017-06-01:
Don't use editor with administrative privileges. PIatformIO doesn't require these permissions.

dude719, 2017-06-01:
I need to on Windows 10 in order to run certain commands within Atom.io, I don't think thi is an issue that should be brushed off due to the fact I am running as administrator. Nonetheless, I guess I'll just uninstall PlatformIO for now until it's more stable on Windows.

ivankravets, 2017-06-01:
Ok, I'll try to reproduce it. What is your Windows version?

dude719, 2017-06-01:
- Windows 10 build 14393
- Python 2.7 32-bit
- List of packages installed other than PlatformIO:
```
atom-material-ui
build
busy
busy-signal
file-icons
git-plus
intentions
language-ini
linter
linter-clang
linter-gcc
linter-ui-default
minimap
```

ivankravets, 2017-06-02:
I have Win10 in virtual machine and can't reproduce this issue.

Could you provide TeamViewer session for debug? me@ikravets.com

Thanks!

commit 3faa24e9 (2017-06-27): Fix issue when can not load broken PIO Core state // Resolve #551 Resolve #564 Resolve #577 Resolve #649
closed 2017-06-27 by ivankravets