platformio/platformio-atom-ide #554: InstallationManager

alxm31, 2017-06-02:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/sk/tpjm_ksd5d79nq8607v1kv0h0000gn/T/tmp-15673wozMFKih2Sb/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/sk/tpjm_ksd5d79nq8607v1kv0h0000gn/T/tmp-15673wozMFKih2Sb/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/sk/tpjm_ksd5d79nq8607v1kv0h0000gn/T/tmp-15673wozMFKih2Sb/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/alex/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/Users/alex/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs

    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-06-02:
Could you contact me@ikravets.com for details? Thanks!

ivankravets, 2017-06-03:
Please provide an output from system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

ivankravets, 2017-06-03:
Could you try 

```
rm -r ~/.platformio/penv

sudo python -c "$(curl -fsSL https://bootstrap.pypa.io/get-pip.py)"
```

Start Atom

closed 2017-06-27 by ivankravets