platformio/platformio-atom-ide #556: InstallationManager

bytesbay, 2017-06-02:
PIP: /home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

Exception:
Traceback (most recent call last):
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 778, in install
    requirement.uninstall(auto_confirm=True)
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 754, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/home/loshap/.platformio/penv/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/usr/lib/python2.7/shutil.py", line 300, in move
    rmtree(src)
  File "/usr/lib/python2.7/shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "/usr/lib/python2.7/shutil.py", line 250, in rmtree
    os.remove(fullname)
OSError: [Errno 13] Permission denied: '/home/loshap/.platformio/penv/lib/python2.7/site-packages/platformio-3.4.0b9-py2.7.egg-info/installed-files.txt'

bytesbay, 2017-06-02:
sorry, missclicked, i already fixed

ivankravets, 2017-06-02:
How did you fix that?

bytesbay, 2017-06-02:
@ivankravets I just had permission problems before with /home dir, i just did (Ubuntu 14.04 LTS)
`sudo chown -R <youruser> /home/<youruser>`
But for most of people - another command
`sudo chown -R <youruser> /home/<youruser>/.platformio`

Кстати, спасибо ребята за ваш продукт ))

ivankravets, 2017-06-02:
Happy coding with PlatformIO! :)

closed 2017-06-02 by ivankravets