platformio/platformio-atom-ide #557: Platform command failed: platform install linux_arm@1.3.0

pierreverbakel, 2017-06-02:
Error: Error: Could not find a version that satisfies the requirement '~1.40802.0' for your system 'windows_amd64'

    at file:///C:/Users/Pierre/.atom/packages/platformio-ide/lib/platform/helpers.js:61:25
    at onExit (file:///C:/Users/Pierre/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/Pierre/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (D:\Users\Pierre\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:322:11)
    at D:\Users\Pierre\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:343:11
    at Socket.<anonymous> (D:\Users\Pierre\AppData\Local\atom\app-1.17.2\resources\app\src\buffered-process.js:204:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)

pierreverbakel, 2017-06-02:
While installing support for wiringPi

ivankravets, 2017-06-03:
Duplicate of https://github.com/platformio/platform-linux_arm/issues/2

closed 2017-06-03 by ivankravets