platformio/platformio-atom-ide #560: Gr eiainErr oe0pnce Itrutwttmoto P0) Error. and broken character

wjm0621, 2017-06-03:
[image]

I'm Using Sparkfun ESP32 Thing. and sending websocket message.
At the shown image, I met Gr eiainErr oe0pnce Itrutwttmoto P0) Error....

It is not always raised. just somtime shown.
I can't find this at google.

What is this Error means??

And some character is broken. (broken character is English character)
At shown image, code line 146. It will print "TCP handshaking with Server Connected. "
But below the Code, at word "handshaking", 'k', 'n' are broke.
How can i fix this problem??

ivankravets, 2017-06-03:
Duplicate of https://github.com/platformio/platform-espressif32/issues/23

ImanolGo, 2017-06-18:
Were you able to solve it? I have exactly the same issue!

wjm0621, 2017-06-18:
I Can't solve it yet..

ImanolGo, 2017-06-20:
I change the board and that solved the issue.

ivankravets, 2017-06-20:
@ImanolGo what was your previous board?

ImanolGo, 2017-06-21:
My previous board was the Sparkfun ESP32 Thing.

Now I changed to the  HIMALAYA ESP32 Dev Board.

ivankravets, 2017-06-27:
Is this a bug with a board?

ImanolGo, 2017-06-28:
Maybe. I tried it in two different Sparkfun ESP32 Thing boards, both with the same results.

ivankravets, 2017-06-28:
Does it work with Arduino IDE serial port monitor?

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298
closed 2017-06-28 by ivankravets