platformio/platformio-atom-ide #567: InstallationManager

MrAmazoni, 2017-06-04:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Windows\TEMP\tmp-2724Tc27QZ5FoRZ2\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Windows\TEMP\tmp-2724Tc27QZ5FoRZ2\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Windows\TEMP\tmp-2724Tc27QZ5FoRZ2\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Windows\TEMP\tmp-2724Tc27QZ5FoRZ2\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Windows\TEMP\tmp-2724Tc27QZ5FoRZ2\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5]

ivankravets, 2017-06-04:
Do you use Antivirus? Try to disable it for a while.

MrAmazoni, 2017-06-04:
I'm using windows defender.I don't think it affect this process.

ivankravets, 2017-06-05:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

MrAmazoni, 2017-06-05:
I have fixed that prolem with reinstalling platform io module but I have found this problem 
[image]

MrAmazoni, 2017-06-05:
[image]

MrAmazoni, 2017-06-05:
$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
adodbapi (2.6.0.7)
alabaster (0.7.9)
algopy (0.5.3)
altair (1.2.0)
args (0.1.0)
astroid (1.4.9)
astroML (0.3)
Babel (2.3.4)
backports-abc (0.5)
baresql (0.7.4)
bcolz (1.1.0)
beautifulsoup4 (4.5.3)
blaze (0.10.1)
bleach (1.5.0)
blosc (1.4.4)
bloscpack (0.11.0)
bokeh (0.12.4)
boto3 (1.4.1)
botocore (1.4.36)
Bottleneck (1.2.0)
bqplot (0.8.4)
brewer2mpl (1.4.1)
certifi (2016.9.26)
cffi (1.9.1)
chardet (2.3.0)
click (6.7)
clint (0.5.1)
cloudpickle (0.2.2)
clrmagic (0.0.1a2)
colorama (0.3.7)
CommonMark (0.5.4)
cvxopt (1.1.9)
cx-Freeze (5.0.1)
cycler (0.10.0)
cyordereddict (1.0.0)
Cython (0.25.2)
cytoolz (0.8.2)
dask (0.13.0)
datashape (0.5.2)
db.py (0.5.2)
decorator (4.0.11)
distributed (1.15.2)
docopt (0.6.2)
docrepr (0.1.0)
docutils (0.13.1)
emcee (2.2.1)
entrypoints (0.2.2)
enum34 (1.1.6)
fasteners (0.14.1)
Flask (0.12)
Flask-Cors (2.1.2)
formlayout (1.1.0)
fuzzywuzzy (0.14.0)
geopy (1.11.0)
ggplot (0.11.5)
greenlet (0.4.11)
guidata (1.7.6)
guiqwt (3.0.3)
h5py (2.6.0)
HeapDict (1.0.0)
holoviews (1.7.dev4)
html5lib (0.9999999)
husl (4.0.3)
imageio (2.1.1)
imagesize (0.7.1)
ipykernel (4.5.2)
ipyleaflet (0.2.1)
ipyparallel (6.0.0)
ipython (5.1.0)
ipython-genutils (0.1.0)
ipython-sql (0.3.8)
ipywidgets (5.2.2)
isort (4.2.5)
itsdangerous (0.24)
jedi (0.9.0)
Jinja2 (2.9.4)
jmespath (0.9.0)
joblib (0.10.3)
jsonschema (2.5.1)
julia (0.1.1)
jupyter (1.0.0)
jupyter-client (4.4.0)
jupyter-console (5.0.0)
jupyter-core (4.2.1)
Keras (1.2.1)
Lasagne (0.2.dev1)
lazy-object-proxy (1.2.2)
llvmlite (0.15.0)
lmfit (0.9.5)
locket (0.2.0)
lxml (3.7.2)
Markdown (2.6.7)
MarkupSafe (1.0.dev0)
matlabengineforpython (R2016a)
matplotlib (2.0.0)
mccabe (0.5.3)
metakernel (0.18.2)
mistune (0.7.3)
mkl-service (1.1.2)
monotonic (1.2)
mpld3 (0.3)
mpmath (0.19)
msgpack-python (0.4.8)
multipledispatch (0.4.9)
mysql-connector-python (2.0.4)
nbconvert (5.1.1)
nbconvert-reportlab (0.1)
nbdime (0.1.2)
nbformat (4.2.0)
nbsphinx (0.2.12)
netCDF4 (1.2.7)
networkx (1.11)
nltk (3.2.2)
nose (1.3.7)
notebook (4.3.1)
numba (0.30.1)
numdifftools (0.9.20)
numexpr (2.6.1)
numpy (1.11.3+mkl)
numpydoc (0.6.0)
oct2py (3.8.1)
octave-kernel (0.23.2)
odo (0.5.0)
olefile (0.44)
pandas (0.19.2)
pandas-datareader (0.3.0.post0)
pandocfilters (1.4.1)
param (1.4.2)
partd (0.3.7)
patsy (0.4.1)
pdfrw (0.3)
pep8 (1.7.0)
pexpect (4.2.1)
pg8000 (1.10.5)
pickleshare (0.7.4)
Pillow (4.0.0)
pip (9.0.1)
pkginfo (1.4.1)
prettytable (0.7.2)
prettytensor (0.7.3)
prompt-toolkit (1.0.9)
protobuf (3.1.0)
psutil (5.0.1)
ptpython (0.36)
ptyprocess (0.5.1)
PuLP (1.6.1)
py (1.4.32)
PyAudio (0.2.10)
pybars3 (0.9.2)
pycodestyle (2.2.0)
pycparser (2.17)
pyflakes (1.5.0)
pyflux (0.4.14)
Pygments (2.1.3)
pylint (1.6.5)
pymc (2.3.6)
pymc3 (3.0)
PyMeta3 (0.5.1)
pymongo (3.4.0)
pyodbc (4.0.3)
PyOpenGL (3.1.1)
pypandoc (1.3.2)
pyparsing (2.1.10)
PyQt5 (5.7.1)
pyqtgraph (0.10.0)
pyserial (3.2.1)
pystache (0.5.4)
pytest (3.0.5)
python-dateutil (2.6.0)
python-hdf4 (0.9)
python-Levenshtein (0.12.0)
pythonnet (2.2.1)
PythonQwt (0.5.5)
pytz (2016.10)
pywin32 (220.1)
PyYAML (3.12)
pyzmq (16.0.2)
QtAwesome (0.4.3)
qtconsole (4.2.1)
QtPy (1.2.1)
recommonmark (0.4.0)
redis (2.10.5)
reportlab (3.3.0)
requests (2.12.5)
requests-file (1.4.1)
requests-ftp (0.3.1)
requests-toolbelt (0.7.0)
rope-py3k (0.9.4.post1)
rpy2 (2.8.5)
Rx (1.5.7)
s3fs (0.0.8)
s3transfer (0.1.10)
scikit-fuzzy (0.3.dev0)
scikit-image (0.12.3)
scikit-learn (0.18.1)
scikit-neuralnetwork (0.7)
scikit-optimize (0.2)
scilab2py (0.6)
scipy (0.18.1)
seaborn (0.8.dev0)
setuptools (33.1.1)
simplegeneric (0.8.1)
simplejson (3.10.0)
sip (4.19)
six (1.10.0)
snakeviz (0.4.1)
snowballstemmer (1.2.1)
sortedcollections (0.4.2)
sortedcontainers (1.5.7)
sounddevice (0.3.6)
speedtest-cli (1.0.6)
Sphinx (1.5.1)
sphinx-rtd-theme (0.1.9)
spyder (3.1.2)
SQLAlchemy (1.1.5)
sqlite-bro (0.8.11)
sqlparse (0.2.2)
statsmodels (0.8.0rc1)
sympy (1.0)
tables (3.3.0)
tblib (1.3.0)
tensorflow (0.12.1)
testpath (0.3)
Theano (0.8.2)
toolz (0.8.2)
tornado (4.4.2)
tqdm (4.11.1)
traitlets (4.3.1)
traittypes (0.0.6)
twine (1.8.1)
twitter (1.17.1)
vega (0.4.4)
wcwidth (0.1.7)
Werkzeug (0.11.15)
wheel (0.29.0)
widgetsnbextension (1.2.6)
win-unicode-console (0.5)
winpython (1.7.20170128)
wordcloud (1.2.1)
wrapt (1.10.8)
xarray (0.9.0)
xlrd (1.0.0)
XlsxWriter (0.9.5)
zarr (2.1.4)
zict (0.1.1)

baileysage, 2017-06-05:
@MrAmazoni How did you fix the original issue? I'm having the same problem.

MrAmazoni, 2017-06-05:
reinstall the platform io plugin for atom and it will fix it

baileysage, 2017-06-06:
@MrAmazoni Crap. That didn't work for me.....

closed 2017-06-27 by ivankravets