platformio/platformio-atom-ide #569: InstallationManager

celindho, 2017-06-05:
Error: Error: Failed to download file https://github.com/platformio/platformio-examples/tarball/master: 
    at ProjectExamplesStage.<anonymous> (/Users/clindholm/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (/Users/clindholm/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:13:273)

ivankravets, 2017-06-05:
Could you restart Atom? Do you see this error again?

celindho, 2017-06-05:
Hi,

Thanks for the fast reply! I restarted Atom and then it installed the missing components. Now everything appears to work as expected.

BR,
-Christian
-- -- -- -- -- -- -- --
Christian Lindholm
christian.lindholm@iki.fi <mailto:christian.lindholm@iki.fi>

tel. +358 (0) 50 371 5459
Lysmaskgränden 4, 02400 Kyrkslätt

ivankravets, 2017-06-05:
Thanks a lot for confirmation

closed 2017-06-05 by ivankravets