platformio/platformio-atom-ide #581: InstallationManager

philippetheis, 2017-06-06:
PIP: Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\theis\.platformio\penv\Scripts\pip.exe\__main__.py", line 5, in <module>
  File "c:\users\theis\.platformio\penv\lib\site-packages\pip\__init__.py", line 21, in <module>
    from pip._vendor.requests.packages.urllib3.exceptions import DependencyWarning
  File "c:\users\theis\.platformio\penv\lib\site-packages\pip\_vendor\requests\__init__.py", line 62, in <module>
    from .packages.urllib3.exceptions import DependencyWarning
  File "c:\users\theis\.platformio\penv\lib\site-packages\pip\_vendor\requests\packages\__init__.py", line 29, in <module>
    import urllib3
ImportError: No module named urllib3

ivankravets, 2017-06-06:
Please remove `c:\users\theis.platformio\penv`

Also, if you have `requests` installed in a system, please
```
pip install -U requests
```

closed 2017-06-06 by ivankravets

philippetheis, 2017-06-06:
thank you @ivankravets for your fast response.
the problem was a wrong env variable in the windows path variable.
best regards