platformio/platformio-atom-ide #583: C/C++ project index rebuild failed

abhisheksafui, 2017-06-06:
Error: Detected unknown package 'espressif8266_stage'
I installed staging using command:
```

platformio platform install https://github.com/platformio/platform-espressif8266.git
```
Here is the output:

```
PlatformManager: Installing platform-espressif8266
espressif8266 @ e75d211 is already installed
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
toolchain-xtensa @ 1.40802.0 is already installed
PackageManager: Installing tool-esptool @ ~1.409.0
tool-esptool @ 1.409.0 is already installed
The platform 'https://github.com/platformio/platform-espressif8266.git' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
```

but still getting the error. Where I am going wrong?

abhisheksafui, 2017-06-06:
I have updated arduino core using: 
```
platformio platform update https://github.com/platformio/platform-espressif8266.git
Platform https://github.com/platformio/platform-espressif8266.git
--------
Updating espressif8266                   @ e75d211        [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20300.1      [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```

abhisheksafui, 2017-06-06:
Build Output:

```
[Tue Jun  6 23:49:39 2017] Processing nodemcuv2 (platform: espressif8266_stage; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------
PlatformManager: Installing espressif8266_stage
Error: Detected unknown package 'espressif8266_stage'
 
```

abhisheksafui, 2017-06-06:
I started project containing a single file main.cpp
```

#include <Arduino.h>
#include<ESP8266WiFi.h>
#include<ESP8266mDNS.h>
#include <ESPAsyncUDP.h>

void setup(){

}

void loop(){

}

```

abhisheksafui, 2017-06-07:
platformio platform install https://github.com/platformio/platform-espressif8266.git**#feature/stage**

I missed **#feature/stage** , I thought it was comment. 
Solved

closed 2017-06-07 by abhisheksafui

alexiszecharies, 2017-12-03:
How did you fix this. I have the same problem and done the same. The last thing  you posted is the same as the first in which it was not working