platformio/platformio-atom-ide #588: C/C++ project index rebuild failed

ilovetogetspamed, 2017-06-07:
```Exception: [Errno 13] Permission denied: 'monitor/.pioenvs/do-not-modify-files-here.url':
File "/Users/********/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 87:

PROJECTPIOENVS_DIR=util.get_projectpioenvs_dir(),
File "/Users/*******/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 309:
raise Exception(e)

========================== [ERROR] Took 0.40 seconds ==========================
```

```
ls: monitor/.pioenvs/do-not-modify-files-here.url: Permission denied
MacBook-Pro:~ *******$ sudo ls -l monitor/.pioenvs/do-not-modify-files-here.url
Password:
-rw-r--r--  1 *******  staff  82 Jun  6 16:00 monitor/.pioenvs/do-not-modify-files-here.url
```

Happens every time I start Atom now.  Or if i select Rebuild C/C++ Project Index (Autocomplete, Linter) from the PlatformIO Menu.

IDE 2.0.0-beta.6/ Core 3.4.0b10
autocomplete-clang0.11.3
linter 2.1.4
linter-gcc 0.7.1
linter-ui-default 1.6.0

ilovetogetspamed, 2017-06-07:
OS X El Capitan Version 10.11.6 on MacBook Pro (Retina, 13-inch, Early 2015)

ivankravets, 2017-06-27:
Please remove manually this folder `..monitor/.pioenvs` in your project

closed 2017-06-27 by ivankravets