platformio/platformio-atom-ide #598: InstallationManager - Windows Error 5

baileysage, 2017-06-08:
I have given the Python27 folder and everything in it full access permissions, and I get this error 5 still. Uninstalling Platformio, deleting the .platformio folder, turning off virus scan, and reinstalling did not help.

Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-2476W2RnOdAMaKNz\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-2476W2RnOdAMaKNz\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-2476W2RnOdAMaKNz\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-2476W2RnOdAMaKNz\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\steinfba\AppData\Local\Temp\tmp-2476W2RnOdAMaKNz\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

baileysage, 2017-06-08:
And yes, I tried apm install platformio/platformio-atom-ide as well.

ivankravets, 2017-06-28:
Fixed in https://github.com/platformio/platformio-atom-ide/commit/ac7dbefea94f7009d795d2a38e2c4aadfb63c080

closed 2017-06-28 by ivankravets