platformio/platformio-atom-ide #607: Library command failed: lib --global stats --json-output

sanny1, 2017-06-10:
Error: Error: spawn platformio ENOENT
    at utils.runPIOCommand.spawnOptions (/Users/sanyogchhetri/.atom/packages/platformio-ide/lib/library/helpers.js:39:25)
    at onExit (/Users/sanyogchhetri/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at /Users/sanyogchhetri/.atom/packages/platformio-ide/lib/utils.js:107:5
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:1)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:1)
    at BufferedProcess.handleError (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:385:1)
    at ChildProcess.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:373:1)
    at emitOne (events.js:96:13)
    at ChildProcess.emit (events.js:188:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:213:12)
    at onErrorNT (internal/child_process.js:359:16)
    at _combinedTickCallback (internal/process/next_tick.js:74:11)
    at process._tickCallback (internal/process/next_tick.js:98:9)

ivankravets, 2017-06-10:
http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command

sanny1, 2017-06-10:
Sorry I just needed to update it, Haven't had enough sleep :P, Thank you for the fast reply

closed 2017-06-10 by sanny1