platformio/platformio-atom-ide #613: Uncaught SyntaxError: Unexpected token C in JSON at position 0

lightway29, 2017-06-12:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.17.2 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-beta.6

### Stack Trace

Uncaught SyntaxError: Unexpected token C in JSON at position 0

```
At file:///Applications/Atom.app/Contents/Resources/app/static/index.html:1

SyntaxError: Unexpected token C in JSON at position 0
    at Object.parse (native)
    at /packages/platformio-ide/lib/account/helpers.js:29:55
    at onExit (/packages/platformio-ide/lib/utils.js:92:7)
    at exit (/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:1)
    at ChildProcess.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:368:1)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
```

### Non-Core Packages

```
atom-beautify 0.29.26 
autocomplete-clang 0.11.3 
build 0.68.0 
build-platformio 1.2.2 
busy 0.7.0 
busy-signal 1.4.3 
clipboard-history 0.6.6 
file-icons 2.1.7 
intentions 1.1.2 
language-ini 1.19.0 
linter 2.1.4 
linter-gcc 0.7.1 
linter-ui-default 1.6.0 
minimap 4.28.2 
multi-cursor 2.1.5 
platformio-ide 2.0.0-beta.6 
platformio-ide-debugger 1.2.2 
platformio-ide-terminal 2.5.1 
standard-formatter 2.8.0 
tool-bar 1.1.0 
```

ivankravets, 2017-06-27:
Please restart Atom

ivankravets, 2017-06-27:
Also, try to reinstall PIO IDE http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

closed 2017-06-27 by ivankravets