platformio/platformio-atom-ide #622: InstallationManager fails on PIO update

bigwoof, 2017-06-14:
This has started happening after my laptop unexpectedly rebooted during an update.  Now the POI update crashes when initialising POI core.

I am running linux (Bunsenlabs - based on Debian stable).

I have tried to completely remove and reinstall atom and POI I have even tried another IDE and still no luck.

any help would be appreciated.

Here is the log:

Virtualenv Download: Traceback (most recent call last):
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv.py", line 843, in install_wheel

    cert_data = pkgutil.get_data("pip._vendor.requests", "cacert.pem")
  File "/usr/lib/python2.7/pkgutil.py", line 578, in get_data

    loader = get_loader(package)
  File "/usr/lib/python2.7/pkgutil.py", line 464, in get_loader

    return find_loader(fullname)
  File "/usr/lib/python2.7/pkgutil.py", line 474, in find_loader

    for importer in iter_importers(fullname):
  File "/usr/lib/python2.7/pkgutil.py", line 430, in iter_importers

    __import__(pkg)
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/__init__.py", line 13, in <module>

  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/utils/__init__.py", line 26, in <module>
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3130, in <module>
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3116, in _call_aside
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3143, in _initialize_master_working_set
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 633, in _build_master

  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 626, in __init__
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 682, in add_entry
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2120, in find_on_path
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2510, in from_location

  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2827, in _reload_version
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2475, in _version_from_file
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2643, in _get_metadata
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2011, in get_metadata_lines
  File "/tmp/tmp-27537qXMLy75gGmk/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2006, in get_metadata

  File "/usr/lib/python2.7/codecs.py", line 296, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)

UnicodeDecodeError: 'utf8' codec can't decode byte 0xb6 in position 147: invalid start byte

ivankravets, 2017-06-27:
Please provide an output from system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

bigwoof, 2017-06-29:
Hi Ivan, 

Thanks for the help, here is the output,  It looks like some errors are showing up but unfortunately i have no idea how to solve them.

PS: no urgency on this as I will be out of the office for a couple of weeks and will not have access to this machine I will try to troubleshoot as soon as I get back. 

Cheers

Phil
  
```
echo $PATH
/home/philip/bin:/usr/sbin:/sbin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
```
```
python --version
Python 2.7.9

```
```
pip --version
/usr/local/lib/python2.7/dist-packages/pip/_vendor/pkg_resources/__init__.py:1874: UserWarning: /usr/lib/pymodules/python2.7/rpl-1.5.5.egg-info could not be properly decoded in UTF-8
  warnings.warn(msg)
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

```
```
pip list
/usr/local/lib/python2.7/dist-packages/pip/_vendor/pkg_resources/__init__.py:1874: UserWarning: /usr/lib/pymodules/python2.7/rpl-1.5.5.egg-info could not be properly decoded in UTF-8
  warnings.warn(msg)
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
apt-xapian-index (0.47)
arandr (0.1.7.1)
arrow (0.10.0)
bottle (0.12.13)
certifi (2017.4.17)
chardet (3.0.3)
click (5.1)
colorama (0.3.9)
ecdsa (0.11)
galternatives (0.9)
idna (2.5)
lockfile (0.12.2)
lxml (3.4.0)
Mirage (0.9.5.1)
numpy (1.8.2)
obMenu (1.0)
paramiko (1.15.1)
pexpect (3.2)
Pillow (2.6.1)
pip (9.0.1)
platformio (3.3.1)
pycrypto (2.6.1)
pycups (1.9.63)
pycurl (7.19.5)
pygobject (3.14.0)
pygpgme (0.3)
pyserial (3.3)
pysmbc (1.0.15.3)
python-apt (0.9.3.12)
python-dateutil (2.6.0)
python-debian (0.1.27)
python-xlib (0.14)
pyxdg (0.25)
reportlab (3.1.8)
requests (2.17.3)
rpl (1.5.5)
semantic-version (2.6.0)
setuptools (36.0.1)
six (1.10.0)
urllib3 (1.21.1)
wheel (0.29.0)

```

ivankravets, 2017-06-29:
Please open system terminal and remove duplicated PIO Core

```
sudo pip uninstall platformio
rm -rf ~/.platformio/penv
```

Restart Atom

bigwoof, 2017-06-29:
OK I ran the commands but i am still getting errors when atom starts:

I am happy completely purge platformio and start fresh as I have backed up all the files I need.  I havent managed to do a clean install with everything I have tried though.

```
Virtualenv Create: Traceback (most recent call last):
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment
    download=download,
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv.py", line 843, in install_wheel
    cert_data = pkgutil.get_data("pip._vendor.requests", "cacert.pem")
  File "/usr/lib/python2.7/pkgutil.py", line 578, in get_data
    loader = get_loader(package)
  File "/usr/lib/python2.7/pkgutil.py", line 464, in get_loader
    return find_loader(fullname)
  File "/usr/lib/python2.7/pkgutil.py", line 474, in find_loader
    for importer in iter_importers(fullname):
  File "/usr/lib/python2.7/pkgutil.py", line 430, in iter_importers
    __import__(pkg)
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/__init__.py", line 13, in <module>
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/utils/__init__.py", line 26, in <module>
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3130, in <module>
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3116, in _call_aside
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3143, in _initialize_master_working_set
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 633, in _build_master
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 626, in __init__
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 682, in add_entry
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2120, in find_on_path
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2510, in from_location
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2827, in _reload_version
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2475, in _version_from_file
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2643, in _get_metadata
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2011, in get_metadata_lines
  File "/home/philip/.platformio/.cache-ide/tmp-2271PJB6fcVYPJLB/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2006, in get_metadata
  File "/usr/lib/python2.7/codecs.py", line 296, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xb6 in position 147: invalid start byte
```

ivankravets, 2017-06-29:
Please open system terminal and run:

```
sudo pip install virtualenv
```

Restart Atom

bigwoof, 2017-07-11:
Hi Ivan,

Looks like that solved the problem, sorry for responding so late (just got back into town yesterday).

ivankravets, 2017-07-11:
Happy coding with PlatformIO!

closed 2017-07-11 by ivankravets