platformio/platformio-atom-ide #630: InstallationManager

suchakra, 2017-06-16:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/xz/lz759kdd34b1_1_yzpv0ssz40000gn/T/tmp-59349qtN4oHTJ97f0/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/xz/lz759kdd34b1_1_yzpv0ssz40000gn/T/tmp-59349qtN4oHTJ97f0/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)

  File "/var/folders/xz/lz759kdd34b1_1_yzpv0ssz40000gn/T/tmp-59349qtN4oHTJ97f0/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/var/folders/xz/lz759kdd34b1_1_yzpv0ssz40000gn/T/tmp-59349qtN4oHTJ97f0/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/var/folders/xz/lz759kdd34b1_1_yzpv0ssz40000gn/T/tmp-59349qtN4oHTJ97f0/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))
OSError: Command /Users/SUMABTCH/.platformio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-06-27:
Please provide an output from system Terminal 
```
echo $PATH
python --version
pip --version
pip list
```

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298