platformio/platformio-atom-ide #639: InstallationManager

embedded-dev, 2017-06-20:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/by/9_kmhl791wl_5d304lvs7xwr00hc1v/T/tmp-490202JUE20kVyvh3/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/by/9_kmhl791wl_5d304lvs7xwr00hc1v/T/tmp-490202JUE20kVyvh3/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/by/9_kmhl791wl_5d304lvs7xwr00hc1v/T/tmp-490202JUE20kVyvh3/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/var/folders/by/9_kmhl791wl_5d304lvs7xwr00hc1v/T/tmp-490202JUE20kVyvh3/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/var/folders/by/9_kmhl791wl_5d304lvs7xwr00hc1v/T/tmp-490202JUE20kVyvh3/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))
OSError: Command /Users/jandre/.platformio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-06-27:
1. What is your macOS version?
2. Please open system Terminal and provide an output of 
```
echo $PATH
python --version
pip --version
pip list
```

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

embedded-dev, 2017-06-28:
I'm confused, I had this issue but I don't remember opening this issue.  I restarted the install and it completed the second time.  Everything is working correctly; much better editor than Arduino's.  :-)

Thanks for closing.

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298