platformio/platformio-atom-ide #644: InstallationManager

milestorm, 2017-06-22:
Virtualenv Download: Traceback (most recent call last):
  File "/var/folders/_n/8jkf5czn20b4mxx57m23q7p1_n269r/T/tmp-4645E6olUjRlxmxp/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/var/folders/_n/8jkf5czn20b4mxx57m23q7p1_n269r/T/tmp-4645E6olUjRlxmxp/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/var/folders/_n/8jkf5czn20b4mxx57m23q7p1_n269r/T/tmp-4645E6olUjRlxmxp/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/var/folders/_n/8jkf5czn20b4mxx57m23q7p1_n269r/T/tmp-4645E6olUjRlxmxp/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)

  File "/var/folders/_n/8jkf5czn20b4mxx57m23q7p1_n269r/T/tmp-4645E6olUjRlxmxp/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))

OSError: Command /Users/ms/...rmio/penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

milestorm, 2017-06-22:
I've tried removing ~/.platformio directory and installing the newest versions of python 2 and 3... and nothing helps. Still the same error.

ivankravets, 2017-06-27:
Please try `sudo pip install -U setuptools` in system Terminal and restart Atom

milestorm, 2017-06-28:
I've tried your command and gets me still the same error :(

ivankravets, 2017-06-28:
Please provide an output from system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

milestorm, 2017-06-28:
`➜  ~ echo $PATH
/Library/Frameworks/Python.framework/Versions/2.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/munki
➜  ~ python --version
Python 2.7.13
➜  ~ pip --version
pip 9.0.1 from /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)
➜  ~ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (36.0.1)`

closed 2017-06-28 by ivankravets

ivankravets, 2017-06-28:
1. Please open Atom > Settings > Packages > PlatformIO IDE
2. Check "Advanced > Use development version of PlatformIO IDE"
3. Wait while Atom will re-install PIO IDE

Does it work now?

milestorm, 2017-06-28:
GREAT! It works! :) Thank you very much for your support, have a nice day ;)

ivankravets, 2017-06-28:
Thank you that reported!

Happy coding with PlatformIO!

commit ac7dbefe (2017-06-28): Don't use system temporary directory for virtualenv tool // Resolve #644 Resolve #639 Resolve #630 Resolve #560 Resolve #498 Resolve #492 Resolve #475 Resolve #298