platformio/platformio-atom-ide #664: InstallationManager

ameisso, 2017-06-26:
I got this error while trying to install beta 2.0.6. 

Looks like it's stuck in downloading the project examples. 
Rolling back to older version for the moment. 
Any help apreciated.

Regards 

Virtualenv Download: Error: Failed to download file https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz: null

ivankravets, 2017-06-27:
What is your OS?

ameisso, 2017-06-28:
OSX 10.11.6 

Don't know if it's related but I had [this issue](https://github.com/platformio/platformio-atom-ide/issues/313#issuecomment-298711459) with a previous beta

ivankravets, 2017-06-28:
Do you use proxy?

ameisso, 2017-06-28:
Nope, 

I've tried on both my work connection, home connection and phone hotspot connection. None of them have a proxy configured and I get the same error.

Regards

ivankravets, 2017-06-28:
Please provide an output from system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

ameisso, 2017-06-28:
echo $PATH : 
`/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/Applications/Wireshark.app/Contents/MacOS`

python --version
`Python 2.7.13`

pip --version
`pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)`

pip list
`DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
appdirs (1.4.3)
arrow (0.10.0)
bottle (0.12.13)
click (5.1)
colorama (0.3.8)
lockfile (0.12.2)
packaging (16.8)
pip (9.0.1)
pyparsing (2.2.0)
pyserial (3.3)
python-dateutil (2.6.0)
requests (2.13.0)
semantic-version (2.6.0)
setuptools (35.0.1)
six (1.10.0)
wheel (0.29.0)
`

ivankravets, 2017-06-28:
Try to run `pip install virtualenv` and restart Atom

ameisso, 2017-06-28:
it shows that 'PlatformIO' core has been installed (wich was not the case before) 

here is the error : 

`Error: Error: Failed to download file https://github.com/platformio/platformio-examples/tarball/master: null
    at ProjectExamplesStage.<anonymous> (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:148:13)
    at throw (<anonymous>)
    at step (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)`

ivankravets, 2017-06-28:
What is with your machine? No Internet for Atom.

ameisso, 2017-06-28:
I have a MBP 15-inch, Mid 2015. 
The machine is connected to the internet. 
ping 8.8.8.8 answer correctly and I am typing this comment from that computer. 

Thanks for the help. 
Regards

ivankravets, 2017-06-28:
Did you install Atom before PlatformIO?

Could you try to uninstall Atom and make clean installation?

- http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

ameisso, 2017-06-28:
Hello @ivankravets 

i've followed [those](http://platformio.org/get-started/ide?install=atom) install instructions after entering the uninstall lines in my terminal. 

When installing platformIO from atom package manager I have this error : 
`Error: Error: Error: Failed to download file http://dl.platformio.org/ide-data/platformio-examples.tar.gz: null
    at ProjectExamplesStage.<anonymous> (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:158:13)
    at throw (<anonymous>)
    at step (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)`

ivankravets, 2017-06-28:
Please provide an output of `$ apm config list`

ameisso, 2017-06-29:
`; cli configs
globalconfig = "/Users/antoine/.atom/.apm/.apmrc"
user-agent = "npm/3.10.10 node/v6.9.5 darwin x64"
userconfig = "/Users/antoine/.atom/.apmrc"

; environment configs
node-gyp = "/Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp"

; globalconfig /Users/antoine/.atom/.apm/.apmrc
cache = "/Users/antoine/.atom/.apm"
progress = false

; node bin location = /Applications/Atom.app/Contents/Resources/app/apm/bin/node
; cwd = /Users/antoine
; HOME = /Users/antoine/.atom/.node-gyp
; "npm config ls -l" to show all defaults.
`

ivankravets, 2017-06-29:
Could you try [PlatformIO IDE for VSCode](http://docs.platformio.org/page/ide/vscode.html) ? Just for an experiment. Thanks.

ameisso, 2017-06-29:
It's working (install and basic sketch compilation)

ivankravets, 2017-06-29:
Hm... something is wrong with Atom :(

Could you run
```
apm config set strict-ssl false
```

and restart Atom?

ameisso, 2017-06-29:
Hello @ivankravets , 

I've tried this and still got the error : 

`Error: Error: Error: Failed to download file http://dl.platformio.org/ide-data/platformio-examples.tar.gz: null
    at ProjectExamplesStage.<anonymous> (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:158:13)
    at throw (<anonymous>)
    at step (/Users/antoine/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)`

Do you think that an older atom version could help ? If yes wich one ?

ivankravets, 2017-06-29:
> Do you think that an older atom version could help ? If yes wich one ?

Does Atom > Settings > Packags > Search  work?

ameisso, 2017-06-29:
yes, 

see the attached screenshot : 

[image]

ivankravets, 2017-09-06:
Could try the latest 2.0.0-rc.1. Does it work?

ivankravets, 2017-10-17:
Please reopen if you still need help

closed 2017-10-17 by ivankravets

ameanomes, 2017-10-17:
Hello @ivankravets, 

Just installed the last Atom and platformIO version. Everything looks fine. 
Huge coding sprint next week, i'll let you know if I see something. 

Great work by the way ! 

Regards

ameisso, 2017-10-17:
Hello @ivankravets,

Just installed the last Atom and platformIO version. 
Everything looks fine.Huge coding sprint next week, i'll let you know if I see something.

Great work by the way !

Regards

ivankravets, 2017-10-17:
Thanks! Have a nice day! 😊