platformio/platformio-atom-ide #681: InstallationManager

pixelipo, 2017-06-29:
Virtualenv Create: Traceback (most recent call last):
  File "/home/marin/.platformio/.cache-ide/tmp-48859EDDNL31efwq/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/home/marin/.platformio/.cache-ide/tmp-48859EDDNL31efwq/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/home/marin/.platformio/.cache-ide/tmp-48859EDDNL31efwq/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment

    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/marin/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath

    if not isabs(path):
  File "/home/marin/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

$ python --version
Python 2.7.12
$ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)
$ virtualenv --version
15.1.0
$ echo %PATH%
%PATH%

Tried reinstalling, restarting, deleting ~/.platformio folder

Ubuntu 16.04 x64, Atom 1.18

ivankravets, 2017-06-29:
Please remove `/home/marin/.platformio/penv` and restart Atom

pixelipo, 2017-06-29:
Done, still getting the same error, @ivankravets :

Virtualenv Create: Traceback (most recent call last):
  File "/home/marin/.platformio/.cache-ide/tmp-13260AUqpyZD3zhIj/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/home/marin/.platformio/.cache-ide/tmp-13260AUqpyZD3zhIj/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/home/marin/.platformio/.cache-ide/tmp-13260AUqpyZD3zhIj/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/marin/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/home/marin/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-06-30:
Could you provide `$ pip list`?

pixelipo, 2017-07-03:
```
adium-theme-ubuntu (0.3.4)
ansible (2.0.0.2)
backports.ssl-match-hostname (3.4.0.2)
beautifulsoup4 (4.4.1)
cached-property (1.3.0)
chardet (2.3.0)
cryptography (1.2.3)
docker-compose (1.8.0)
docker-py (1.9.0)
dockerpty (0.4.1)
docopt (0.6.2)
ecdsa (0.13)
enum34 (1.1.2)
funcsigs (0.4)
functools32 (3.2.3.post2)
gyp (0.1)
html5lib (0.999)
httplib2 (0.9.1)
idna (2.0)
ipaddress (1.0.16)
Jinja2 (2.8)
jsonschema (2.5.1)
lxml (3.5.0)
MarkupSafe (0.23)
mock (1.3.0)
ndg-httpsclient (0.4.0)
netaddr (0.7.18)
numpy (1.11.0)
paramiko (1.16.0)
pbr (1.8.0)
pip (9.0.1)
pyasn1 (0.1.9)
pycrypto (2.6.1)
pygobject (3.20.0)
pyOpenSSL (0.15.1)
PyYAML (3.11)
requests (2.9.1)
scour (0.32)
setuptools (36.0.1)
six (1.10.0)
texttable (0.8.1)
unity-lens-photos (1.0)
urllib3 (1.13.1)
vboxapi (1.0)
virtualenv (15.1.0)
websocket-client (0.18.0)
wheel (0.29.0)
youtube-dl (2017.5.9)
```

ivankravets, 2017-07-03:
You have `virtualenv` in a system and Atom does not see it. Do you have correct env PATH?

pixelipo, 2017-07-10:
How do I check that, @ivankravets ?

ivankravets, 2017-07-10:
Try this:
```
rm -rf /home/marin/.platformio/penv
virtualenv /home/marin/.platformio/penv
source /home/marin/.platformio/penv/bin/activate
(penv) pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
```

Start Atom

pixelipo, 2017-07-10:
Thanks, that did the trick @ivankravets :beer: 

P.S. Please remove mentions of my home folder name from your comments (I'm a bit paranoid about that sort of things :wink: )

closed 2017-07-10 by pixelipo

ivankravets, 2017-07-10:
Done!

Happy coding with PlatformIO!