platformio/platformio-atom-ide #690: C/C++ project index rebuild failed

sticx, 2017-07-02:
I am quite new to this, but I just freshly installed atom together with platformio on my pc. Wanted to make a new project for my esp32 dev board, got this error. Also looked at other similar issues on here, this link is frequently linked as answer: http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command . However, "Use built-in PIO Core” is enabled. And I don't think I modified the system environment variable path. 

PlatformIO: Failed to initialize PlatformIO project!
Error: Unknown board ID 'esp32doit-devkit-v1'
 [ERROR] Took 0.40 seconds 
C/C++ project index rebuild failed
Error: Unknown board ID 'esp32doit-devkit-v1'
 [ERROR] Took 0.35 seconds

ivankravets, 2017-07-04:
Please navigate to PIO Home > PLatforms and update esp32 dev platform. Try again

closed 2017-07-04 by ivankravets

sticx, 2017-07-08:
Updating the esp32 dev platform to 0.7.5 fixed the issue. Thanks!