platformio/platformio-atom-ide #694: C/C++ project index rebuild failed

rafa400, 2017-07-03:
Error: Unknown board ID 'esp32doit-devkit-v1'

========================== [ERROR] Took 0.38 seconds ==========================
First error said:
PlatformIO: Failed to initialize PlatformIO project!
Second error said:
C/C++ project index rebuild failed

Before this:
I upgraded Atom to the last version on the web and installed 
Then installed platformio-ide and did select a new project using  esp32doit-devkit-v1 board

ivankravets, 2017-07-04:
Please navigate to PIO Home > Platforms > Updates and update all platforms. Start again.

closed 2017-07-04 by ivankravets

rafa400, 2017-07-05:
My system is Devian  jessie
I tried to use the menu option "PlatformIO" / "Update Platforms, Packages and Libraries" and it failed executing "platformio update" command with 
bash: platformio: command not found
(I had installed platformio-ide-terminal)
I searched the command into the install dir and fount (~/.platformio/penv/bin/platformio)
By executing this with the new path I found 
```
cd ~/.platformio
penv/bin/platformio update
```
the update was finally executed and now everything is working with 'esp32doit-devkit-v1'
Thank you

ivankravets, 2017-07-05:
See http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command