platformio/platformio-atom-ide #696: InstallationManager

levonbragg, 2017-07-04:
Error: Error: Error: Failed to download file http://dl.platformio.org/ide-data/platformio-examples.tar.gz: null
    at ProjectExamplesStage.<anonymous> (file:///C:/Users/levon/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:158:13)
    at throw (<anonymous>)
    at step (C:\Users\levon\.atom\packages\platformio-ide\lib\installer\stages\project-examples.js:13:273)
    at process._tickCallback (internal/process/next_tick.js:103:7)

ivankravets, 2017-07-04:
Do you see this issue after Atom restart? Do you use proxy?

levonbragg, 2017-07-05:
I think it may have just been a hiccup…
I have not been able to reproduce the issue.
It has been working perfectly.
No, I do not use a proxy.

Thanks for the follow up.

Levon

closed 2017-07-05 by ivankravets