platformio/platformio-atom-ide #697: C/C++ project index rebuild failed

mariusl, 2017-07-05:
Error: VCS: Unknown repository type ;https://github.com/gioblu/PJON.git

This line is commented out in the platformio.ini file but is still read.

l; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html
[platformio]
env_default = due

; You MUST inject these options into [env:] section
; using ${common_env_data.***} (see below)
[common_env_data]
;build_flags = -D VERSION=1.2.3 -D DEBUG=1
lib_extra_dirs = C:\Arduino_Workspace\arduino\libraries

lib_deps_builtin =
    SPI
    Wire
    Adafruit_GFX
    Adafruit_ILI9340
lib_deps_external =
    ArduinoJson@~5.6,!=5.4
    ;https://github.com/gioblu/PJON.git#v2.0
    ;https://github.com/adafruit/DHT-sensor-library/archive/master.zip

[env:due]
platform = atmelsam
board = due
framework = arduino
lib_deps =
  ${common_env_data.lib_deps_builtin}
  ${common_env_data.lib_deps_external}

[env:sparkfun_samd21_mini_usb]
platform = atmelsam
board = sparkfun_samd21_mini_usb
framework = arduino

ivankravets, 2017-07-05:
if you use `;`, the line should start with `;`, otherwise, use `#` for comment.

closed 2017-07-05 by ivankravets

mariusl, 2017-07-06:
Thanks a lot. Still a bit new here :(