platformio/platformio-atom-ide #703: InstallationManager

Helix70, 2017-07-06:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-4600kg2v8fNm749c\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-4600kg2v8fNm749c\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-4600kg2v8fNm749c\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-4600kg2v8fNm749c\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-4600kg2v8fNm749c\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\Z0035SJY\.p...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

This occurred after removing .platformio directory and re-running atom.

ivankravets, 2017-08-04:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

Helix70, 2017-08-09:
H:\>echo %PATH%
C:\Program Files\Git\cmd;C:\Python27\;C:\Python27\Scripts;C:\Program Files\Commo
n Files\Microsoft Shared\Microsoft Online Services;C:\Program Files (x86)\Common
 Files\Microsoft Shared\Microsoft Online Services;C:\WINDOWS\system32;C:\WINDOWS
;C:\WINDOWS\System32\Wbem;c:\Program Files (x86)\Microsoft SQL Server\100\Tools\
Binn\;c:\Program Files\Microsoft SQL Server\100\Tools\Binn\;c:\Program Files\Mic
rosoft SQL Server\100\DTS\Binn\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;
C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Visual Leak D
etector\bin\Win32;C:\Program Files (x86)\Visual Leak Detector\bin\Win64;C:\Progr
am Files\SlikSvn\bin;C:\MinGW\bin;C:\Program Files (x86)\IVI Foundation\VISA\Win
NT\Bin\;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Found
ation\IVI\bin;C:\Program Files (x86)\MATLAB\MATLAB Compiler Runtime\v716\runtime
\win32;C:\Program Files\Agilent\IO Libraries Suite\bin;C:\Program Files\IVI Foun
dation\VISA\Win64\agvisa;C:\Program Files (x86)\Agilent\IO Libraries Suite\bin;C
:\Program Files (x86)\IVI Foundation\VISA\WinNT\agvisa;C:\Program Files (x86)\No
rdic Semiconductor\nrf5x\bin\;C:\Program Files\Git\cmd;C:\ProgramData\Oracle\Jav
a\javapath;C:\Program Files (x86)\Java\jre1.8.0_111\bin;C:\Program Files (x86)\I
VI Foundation\VISA\WinNT\Bin;C:\Program Files\Microsoft SQL Server\130\Tools\Bin
n\;C:\Program Files\dotnet\;C:\Program Files\LLVM\bin;C:\WINDOWS\System32\Window
sPowerShell\v1.0\;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\QuickT
ime\QTSystem\;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\Z0035SJY\App
Data\Local\atom\bin;C:\Program Files (x86)\Atmel\Flip 3.4.7\bin

H:\>echo %PYTHONPATH%
C:\Program Files (x86)\Keysight\Command Expert\;C:\Program Files (x86)\Keysight\
Command Expert\Addons\python

H:\>python --version
Python 2.7.13

H:\>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

H:\>pip list
DEPRECATION: The default format will switch to columns in the future. You can us
e --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.con
f under the [list] section) to disable this warning.
PIL (1.1.7)
pip (9.0.1)
setuptools (28.8.0)

H:\>

ivankravets, 2017-10-17:
Have you resolved this issue?

Helix70, 2017-10-17:
No, I have stopped using Atom. I can't resolve the proxy issues.

ivankravets, 2017-10-17:
Could you try VSCode?
- http://docs.platformio.org/en/latest/ide/vscode.html

See comparison http://platformio.org/platformio-ide

Helix70, 2017-10-27:
Yes, it works better.

ivankravets, 2017-11-06:
Thanks a lot for a report! We recommend to use VSCode in pair with PlatformIO.

closed 2017-11-06 by ivankravets