platformio/platformio-atom-ide #710: C/C++ project index rebuild failed

manuelbarraza, 2017-07-07:
c:\users\electa04\.platformio\penv\lib\site-packages\platformio\managers\package.py:284: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  elif not pkg_id and manifest['name'] != name:
c:\users\electa04\.platformio\penv\lib\site-packages\platformio\commands\run.py:315: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if any([l.get("name") == lib_name for l in storage['items']]):

Error: Could not find a version that satisfies the requirement '> = 3,24' for your system 'windows_amd64'

ivankravets, 2017-08-04:
Could you provide more details?

ivankravets, 2017-10-17:
Please reopen if you still need help

closed 2017-10-17 by ivankravets

GeorgeFlorian, 2019-03-11:
I have this problem as well. Is there any fix to this ?

ivankravets, 2019-03-11:
@GeorgeFlorian which package do you install?