platformio/platformio-atom-ide #717: InstallationManager

345mu, 2017-07-12:
Virtualenv Create: Traceback (most recent call last):
  File "C:\.platformio\.cache-ide\tmp-7848UYD4iPt6ixkz\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\.platformio\.cache-ide\tmp-7848UYD4iPt6ixkz\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\.platformio\.cache-ide\tmp-7848UYD4iPt6ixkz\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\.platformio\.cache-ide\tmp-7848UYD4iPt6ixkz\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\.platformio\.cache-ide\tmp-7848UYD4iPt6ixkz\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\.platformio\penv\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-10-17:
Please reopen if you still need help

closed 2017-10-17 by ivankravets

me21, 2017-10-27:
It was fixed for me by installing virtualenv package in Python manually.