platformio/platformio-atom-ide #721: C/C++ project index rebuild failed

vknector, 2017-07-13:
D:\SPB_Data\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
val = [x for x in val if x not in dk]

C:\users\lubotec-dc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py:596: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
if path not in self.envorigin.get("CPPPATH", []):

UnicodeDecodeError: Non-ASCII characters found in build environment
See explanation in FAQ > Troubleshooting > Building

http://docs.platformio.org/page/faq.html

 [ERROR] Took 1.03 seconds

ivankravets, 2017-08-04:
Have you resolved this issue?

ivankravets, 2017-09-06:
Please reopen if you need a help

closed 2017-09-06 by ivankravets