platformio/platformio-atom-ide #726: Stop displaying annoying welcome screen to log in

piotrbazan, 2017-07-14:
Whenever atom starts platformio keeps showing screen to log in. Could you remember user option to opt out i.e. skip log in and don't show it in the future?

ivankravets, 2017-07-14:
Settings?

closed 2017-07-14 by ivankravets

piotrbazan, 2017-08-04:
Which type of settings do you mean?
Unchecking 'Show PlatformIO on startup' does not help. The full popup dialog to log in appears. Clicking Skip button is not recorded and the same thing happens after atom restart.

You are basing your platform on open source software, be so kind as to respect users ability to opt-out from logging.

ivankravets, 2017-08-05:
https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L105

We will remove this option and other things soon. Stau tuned with us.