platformio/platformio-atom-ide #749: InstallationManager

Helix70, 2017-07-20:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-5160VJCejdXPqmdN\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-5160VJCejdXPqmdN\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-5160VJCejdXPqmdN\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-5160VJCejdXPqmdN\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Z0035SJY\.platformio\.cache-ide\tmp-5160VJCejdXPqmdN\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\Z0035SJY\.p...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

Helix70, 2017-07-20:
I still can not install platformio at work (Win7Enterprise). I can install at home no problems. I have tried removing the penv directory, same issue. I have uninstalled and reinstalled, same issue. Please help.

mrpraline, 2017-07-27:
Exactly the same error for me, also on Win7Enterprise.

closed 2017-10-17 by ivankravets