platformio/platformio-atom-ide #752: Failed to activate the platformio-ide package - Cannot find module './SyntheticEvent'

Davey3D, 2017-07-21:
Has been working but now when I start Atom I get this error. PlatformIO shows in the menu bar but the options under that heading don't work. Left menubar doesn't appear. Compiling does nothing. Rebooting didn't help.

Win8.1/64. Latest Atom and PlatformIO versions and updates.

Any advice is appreciated!

Thanks

--------------------------------------------
Failed to activate the platformio-ide package
Cannot find module './SyntheticEvent'
Hide Stack Trace
Error: Cannot find module './SyntheticEvent'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\electron.asar\common\reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\module-cache.js:354:58)
    at Module.require (file:///C:/Users/davey/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:40:43)
    at require (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\davey\.atom\packages\platformio-ide\node_modules\react-dom\lib\SyntheticCompositionEvent.js:13:22)
    at Module.get_Module._compile (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/davey/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\davey\.atom\packages\platformio-ide\node_modules\react-dom\lib\BeforeInputEventPlugin.js:16:33)
    at Module.get_Module._compile (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/davey/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\davey\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactDefaultInjection.js:14:30)
    at Module.get_Module._compile (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (file:///C:/Users/davey/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (C:\Users\davey\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\davey\.atom\packages\platformio-ide\node_modules\react-dom\lib\ReactDOM.js:16:29)
The error was thrown from the platformio-ide package. This issue has already been reported.
View Issue

apphead, 2017-07-25:
With the same environment, I got the same error, which was resolved by updating platform-ide to 2.0.0-beta.7

ivankravets, 2017-08-04:
Please uninstall/install PlatformIO IDE package again.

closed 2017-08-04 by ivankravets

Davey3D, 2017-08-05:
I gave up and switched to Visual Studio Code :-/

rdelvax, 2017-08-23:
Just reinstalled atom because I had a LOT of problems, started after an update disabled the terminal. Whatever I tried made it worse, so I reinstalled everything.

Now, when I install the platformio package I get this error... And installing beta.7 apparently isn't an option, because I only get beta.8 to show...

neu-rah, 2017-08-23:
same with beta.8 un/install does not solve it

rdelvax, 2017-08-23:
The only solution that worked for me so far: deleted .atom, .platformio, local/atom and roaming/atom from my user directory, and then copied those folders from an older install on a different computer (same OS...)
but I doubt everyone has a working older installation on a different computer to copy... (maybe copy the files from a backup?)

rainy81, 2017-08-27:
Thank you, rdelvax! Your way worked for me, too:
I deleted rm -r ~/.atom, rm -r ~/.platformio on my Linux Mint 18.2 (Ubuntu 16.04).
Then started Atom again and I was able to install it.

rjaraman21, 2018-04-21:
I am trying to install the platformio-ide in the Atom installer and get errors. First it was "unable to verify first certificate". I downloaded the ide after unzipping and inserted into the Packages Folder.
Now i can see a Platformio drop down but nothing happens. 
I gave up with the Arduino ide due to all the errors and was told Atom/Platform was the way to go.
I cant even get it going. Very frustrating.
Any help would be appreciated.

TD-er, 2018-07-19:
Apparently I just hit the same bug again today.
Just updated the PlatformIO-IDE package in Atom and at startup I get the message it failed to start with a link to this issue.

```
Error: Cannot find module 'tslib'
    at Module._resolveFilename (module.js:543:15)
    at Module._resolveFilename (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\electron.asar\common\reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\module-cache.js:354:58)
    at Module.require (file:///C:/Users/gijs/AppData/Local/atom/app-1.28.2/resources/app.asar/static/index.js:40:43)
    at require (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\gijs\.atom\packages\platformio-ide\node_modules\jsonrpc-lite\jsonrpc.js:7:15)
    at Object.<anonymous> (C:\Users\gijs\.atom\packages\platformio-ide\node_modules\jsonrpc-lite\jsonrpc.js:374:3)
    at Module.get_Module._compile (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///C:/Users/gijs/AppData/Local/atom/app-1.28.2/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:66:33)
    at t (C:\Users\gijs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:355)
    at Object.<anonymous> (C:\Users\gijs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:1173)
    at Object.<anonymous> (C:\Users\gijs\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2:3)
    at Module.get_Module._compile (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///C:/Users/gijs/AppData/Local/atom/app-1.28.2/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\gijs\.atom\packages\platformio-ide\lib\maintenance.js:24:30)
    at Object.<anonymous> (file:///C:/Users/gijs/.atom/packages/platformio-ide/lib/maintenance.js:122:2)
    at Module.get_Module._compile (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\compile-cache.js:240:29)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (file:///C:/Users/gijs/AppData/Local/atom/app-1.28.2/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/gijs/.atom/packages/platformio-ide/lib/main.js:11:19)
    at Object.<anonymous> (file:///C:/Users/gijs/.atom/packages/platformio-ide/lib/main.js:278:22)
    at Module.get_Module._compile (C:\Users\gijs\AppData\Local\atom\app-1.28.2\resources\app\src\native-compile-cache.js:106:36)
```