platformio/platformio-atom-ide #763: CTRL-ALT-U Does not work

Thijxx, 2017-07-24:
In the platformio menu is an option Upload. This option has ALT-CTRL-U behind it, that is the keyboard shortcut.

But platformio does different things when I use the keyboard shortcut or the button in the menu.
The keyboard shortcut is Run and the button is Upload.

ivankravets, 2017-08-04:
I've just tested on my mac and it works. What is your OS?
[image]

Thijxx, 2017-08-05:
Arch Linux

Dreamfire78, 2017-08-06:
Same here. OS: Windows 10
The Shortcut CTRL-ALT-U calls "platformio run" instead of "platformio run --target upload"
The Shortcut in the settings seems correct ctrl-alt-u -> platformio-ide:target:upload
The Button in the menu works.

IDE: 2.0.0-beta.8
Core: 3.5.0a1

mnett0, 2018-08-18:
Hello, 
If you still have the "bug" I found a solution in Atom, just add in the keymap.cson (file --> keymap) : 
`'.platform-linux atom-workspace, .platform-linux atom-text-editor, .platform-win32 atom-workspace, .platform-win32 atom-text-editor':
  'ctrl-alt-u': 'platformio-ide:target:upload'`

Thijxx, 2019-02-01:
This bug is still in there, these days I use Ubuntu and it's the same problem. I cannot access the keymap menu which is not under File by the way, because there is another bug (https://github.com/platformio/platformio-atom-ide/issues/1472). 

Version in 2.9.0