platformio/platformio-atom-ide #764: C/C++ project index rebuild failed

LiaoLi327, 2017-07-25:
C:\Users\13711\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal

val = [x for x in val if x not in dk]

C:\users\13711\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py:596: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
if path not in self.envorigin.get("CPPPATH", []):

UnicodeDecodeError: Non-ASCII characters found in build environment
See explanation in FAQ > Troubleshooting > Building
http://docs.platformio.org/page/faq.html

 [ERROR] Took 507.04 seconds 

Environment d1-debug                	[ERROR]

 [ERROR] Took 507.05 seconds

ivankravets, 2017-08-04:
Do you have non-ASCII chars in project path?

ivankravets, 2017-09-06:
Please reopen if you still need help

closed 2017-09-06 by ivankravets

LiaoLi327, 2017-11-14:
sorry i just see. my project path have non-ASCII chars,It's C:\Lab_project\EspurnaTestMQTT