platformio/platformio-atom-ide #768: C/C++ project index rebuild failed

ricavalcanti, 2017-07-25:
C:\Users\ricjo\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal

val = [x for x in val if x not in dk]

C:\Users\ricjo\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py:596: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
if path not in self.envorigin.get("CPPPATH", []):

UnicodeDecodeError: Non-ASCII characters found in build environment
See explanation in FAQ > Troubleshooting > Building
http://docs.platformio.org/page/faq.html

 [ERROR] Took 1.89 seconds

ivankravets, 2017-08-04:
Do you have non-ASCII chars in project folder path?

ricavalcanti, 2017-08-04:
I fixed that by moving the project folder from google drive folder to desktop

ivankravets, 2017-11-06:
It should be fixed in PIO Core 3.5.0b2. Could you re-test it?

ivankravets, 2017-12-01:
Please reopen if you still need help

JustACatRu, 2019-01-31:
This issue is live again. Kind a.
When using "pio init" with "--ide" parameter (no matter what ide).
I'm talking about PIO CLI install (through "[pip install -U https...](https://docs.platformio.org/en/latest/installation.html#development-version)") last dev version.
Python 3.7.2. All other versions (and error) are visible on screenshot:
[image]
And there is no Non-ASCII characters in my paths as I can see:
[image]

JustACatRu, 2019-02-01:
Forgot to mention: as I understand, this issue does not affect functionality.
I mean, I still can init project and build it without problems (only this error shows up during "init --ide ..." process).
Tested on a project with platform = espressif8266, board = esp12e, framework = arduino.
I will continue testing this bundle (PIO CLI dev + Python 3.7.2 + eclipse as IDE).

JustACatRu, 2019-02-01:
> this issue does not affect functionality.

I was wrong (because was testing on existing projects). **It is affecting functioality.** Actually "pio init --ide ..." does not create all necessary files and folders and new project could not be imported to IDE after that.
So I need manually create (copy from other project) missing files and folders.

JustACatRu, 2019-02-01:
I've made some more tests.
This issue related to espressif8266 platform.
If I select some avr board like "pio init --board pro16MHzatmega328" then "pio init --ide eclipse" works without errors:
[image]
If I select esp board like "pio init --board esp07" then "pio init --ide eclipse" produce error:
[image]
If there is 2 envs in a project and avr goes first in "platformio.ini" then all is workig fine too:
[image]
But if I put "env:esp07" first, "pio init --ide eclipse" stops working again.

@ivankravets, should I transfer this issue to some other branch? As it is not "platformio-atom-ide" related.

JustACatRu, 2019-02-02:
I've modified code a little bit and found traceback of error ([this catch](https://github.com/platformio/platformio-core/blob/e23ca2c10992f10b1906ad616248aa2ef999db23/platformio/builder/main.py#L213) hides traceback):
[image]
Here: https://github.com/platformio/platformio-core/blob/e23ca2c10992f10b1906ad616248aa2ef999db23/platformio/util.py#L420
For me it starts working, if I change that line to any of this variants:
```
result[k] = str(result[k])
result[k] = str(result[k], "windows-1251")
result[k] = result[k].decode("windows-1251")
```
[image]
I do not know if it's a right solution. And which variant is the best and correct. And why it is working with "windows-1251" encoding. My Python knowledge is way too low for this :)
But hope all this will help solve the problem quicker.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets